msm8916-mainline/linux #20: Upstream

Minecrell, 2019-08-07:
### Ready
- [ ] arm64: dts: msm8916-samsung-a2015: Enable wcnss for WiFi and BT (@Minecrell)
  arm64: dts: msm8916-samsung-a5u: Override msm8916 iris compatible

- [ ] arm64: dts: msm8916-longcheer-l8150: Enable wcnss (@TravMurav)

- [ ] arm64: dts: msm8916-samsung-a2015: add gpio-based buttons and a gpio … (@Mis012)

- [ ] arm64: dts: msm8916-longcheer-l8150: Add GPIO Volume Up button (@TravMurav)
  arm64: dts: msm8916-longcheer-l8150: Add Volume Down button on pm8916

- [ ] arm64: dts: msm8916: Add blsp_i2c5 (@Minecrell)
  arm64: dts: msm8916-samsung-a2015: Add touchscreen regulator
  Input: mms114 - add support for mms345l
  arm64: dts: msm8916-samsung-a5u: Add touchscreen

### Pending

### Applied
- [x] [ASoC: qcom: apq8016_sbc: Fix oops with multiple DAI links](https://git.kernel.org/pub/scm/linux/kernel/git/broonie/sound.git/commit/?h=for-5.3&id=8201f11a1f75e3aa7d5327d0b1d8cb544aeaa62f) (5.3)
- [x] [arm64: dts: qcom: Add device tree for Samsung Galaxy A3U/A5U](https://git.kernel.org/pub/scm/linux/kernel/git/qcom/linux.git/commit/?h=arm64-for-5.4&id=1329c1ab0730b521e6cd3051c56a2ff3d55f21e6) (5.4)
- [x] [arm64: dts: qcom: Add device tree for Longcheer L8150](https://git.kernel.org/pub/scm/linux/kernel/git/qcom/linux.git/commit/?h=arm64-for-5.4&id=16e8e8072108426029f0c16dff7fbe77fae3df8f) (5.4)

antony-jr, 2020-11-18:
@Minecrell  I can only see your changes in ```for-next``` branch of ```git://git.kernel.org/pub/scm/linux/kernel/git/qcom/linux.git``` should I submit my patches based on that branch too??

Minecrell, 2020-11-19:
> @Minecrell I can only see your changes in `for-next` branch of `git://git.kernel.org/pub/scm/linux/kernel/git/qcom/linux.git` should I submit my patches based on that branch too??

Yes.

closed 2022-02-12 by stephan-gh