msm8916-mainline/linux #25: Another MSM8916 Device

wojtekt99, 2019-08-23:
Hello im trying to get another msm8916 device on upstream
Its Motorola moto G3 2015 

I found this repo by google when searching for "porting quacomm dsi panels"

So maybe can you mainline also my DTS when i will end it?
(sorry for eng its 02:35AM for me so im not fully rested)

Minecrell, 2019-08-23:
I remember your name, a while ago I tried to figure out what you were doing on `#linux-msm`. I don't think you ever replied, unfortunately ;)

```
23:03 <wojtekt99_> i found msm8916 is as arm64 mach_qcom device and tried to run this compiled kernel but i dont have access to serial on mainboard so i will propably have to find emulator or another phone with msm8916 to debug it
23:22 <lambdadroid> wojtekt99_: What kind of device do you have?
```

If I understand you correctly, the general answer for

> So maybe can you mainline also my DTS when i will end it?

is "no", because we cannot mainline it for you. Only you have the device so you need to do most of the work. However, we can try to assist you and provide lots of tips and tricks from experience we have made when porting our devices.

MSM8916 is a great platform for mainlining, and you can generally expect the device to be quite usable when you are done. It's a bit more difficult without serial, but we might be able to get it working without it based on the other devices.

If you want to go ahead with this, join `#postmarketos-mainline` on [Freenode IRC](https://freenode.net/) or [via Matrix](https://wiki.postmarketos.org/wiki/Matrix_and_IRC) and we can discuss this further.

closed 2019-08-23 by Minecrell

wojtekt99, 2019-08-23:
Yes i was on #linux-msm im not that used to IRC so i wasnt aware somebody answered

I dont mean do work for me but when i done my work just mainline it with other from this repo (easier to me than using mailing lists and others)

I was sleepy so maybe i dont specifed
Im on postmarketos-mainline (maybe kicked for inactivity idk)

Also instead of serial i use kernel pstore it is good method for me
I figured out it is very good method when working on pxa988 device

Minecrell, 2019-08-24:
I don't see you on `#postmarketos-mainline`, so you might need to say something :)

wojtekt99, 2019-08-24:
Idk why it doesnt show me on list but on riot.im matrix client on phone it is listed in my rooms 
https://ibb.co/xSHTspX