msm8916-mainline/linux #190: harpia: Some touchscreens do not work

iyzsong, 2021-04-10:
Original reported in postmarketOS: https://gitlab.com/postmarketOS/pmaports/-/issues/672
> Some harpia devices have different touchscreen revisions that do not seem to work currently. 
> The touch driver seems to be registered, but there is no response from actual touches.

We use this mainline kernel in postmarketOS for harpia, so I think report the issue here maybe more useful..

I have a motorola harpia, lk2nd report panel as `qcom,mdss_dsi_mot_boe_499_720p_video_v1`:
- with kernel version 5.11.7, [config-postmarketos-qcom-msm8916.aarch64](https://gitlab.com/postmarketOS/pmaports/-/blob/master/main/linux-postmarketos-qcom-msm8916/config-postmarketos-qcom-msm8916.aarch64), rmi4_i2c reported `Device was reset during configuration process, status: 0x1!`, then no synaptics input event device available in the system.
- I then modified `rmi_f01_probe`:
```
diff --git a/drivers/input/rmi4/rmi_f01.c b/drivers/input/rmi4/rmi_f01.c
index d7603c50f864..baae1fb50258 100644
--- a/drivers/input/rmi4/rmi_f01.c
+++ b/drivers/input/rmi4/rmi_f01.c
@@ -561,10 +561,9 @@ static int rmi_f01_probe(struct rmi_function *fn)
        }
 
        if (RMI_F01_STATUS_UNCONFIGURED(device_status)) {
-               dev_err(&fn->dev,
+               dev_warn(&fn->dev,
                        "Device was reset during configuration process, status: %#02x!\n",
                        RMI_F01_STATUS_CODE(device_status));
-               return -EINVAL;
        }
 
        dev_set_drvdata(&fn->dev, f01);
```
synaptics input device available, and it can "moving" but not "touch", with rmi_core debug_flags=15, I dumped dmesg:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd030]
[    0.000000] Linux version 5.11.7-postmarketos-qcom-msm8916 (pmos@loto) (aarch64-alpine-linux-musl-gcc (Alpine 10.2.1_git20210328) 10.2.1 20210328, GNU ld (GNU Binutils) 2.35.2) #2 SMP PREEMPT Thu Apr 8 12:34:31 UTC 2021
[    0.000000] Machine model: Motorola Moto G4 Play (harpia)
[    0.000000] earlycon: msm_serial_dm0 at MMIO 0x00000000078af000 (options '')
[    0.000000] printk: bootconsole [msm_serial_dm0] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x0000000085ffffff]
[    0.000000]   node   0: [mem 0x0000000086000000-0x00000000867fffff]
[    0.000000]   node   0: [mem 0x0000000086800000-0x000000008befffff]
[    0.000000]   node   0: [mem 0x000000008bf00000-0x000000008cafffff]
[    0.000000]   node   0: [mem 0x000000008cb00000-0x000000008e9fffff]
[    0.000000]   node   0: [mem 0x000000008ea00000-0x000000008eafffff]
[    0.000000]   node   0: [mem 0x000000008eb00000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000] On node 0 totalpages: 524288
[    0.000000]   DMA zone: 8192 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 524288 pages, LIFO batch:63
[    0.000000] cma: Reserved 32 MiB at 0x00000000fb000000
[    0.000000] percpu: Embedded 22 pages/cpu s49688 r8192 d32232 u90112
[    0.000000] pcpu-alloc: s49688 r8192 d32232 u90112 alloc=22*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM errata 826319, 827319, 824069, or 819472
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516096
[    0.000000] Kernel command line: earlycon console=ttyMSM0,115200 PMOS_NO_OUTPUT_REDIRECT
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1986832K/2097152K available (7872K kernel code, 1122K rwdata, 2884K rodata, 1536K init, 344K bss, 77552K reserved, 32768K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[... 555 lines cut ...]
[  202.557497] rmi4_i2c 1-0020: read 1 bytes at 0x0014: 0 (04)
[  202.562906] rmi4_i2c 1-0020: read 53 bytes at 0x0015: 0 (55 01 00 18 46 eb 03 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)
[  202.562948] input input5: rmi_2d_sensor_abs_report: obj[0]: type: 0x01 X: 395 Y: 1134 Z: 3 WX: 3 WY: 0
[  202.562967] input input5: rmi_2d_sensor_abs_report: obj[1]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0
[  202.562984] input input5: rmi_2d_sensor_abs_report: obj[2]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0
[  202.563000] input input5: rmi_2d_sensor_abs_report: obj[3]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0
[  202.563017] input input5: rmi_2d_sensor_abs_report: obj[4]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0
[  202.568641] rmi4_i2c 1-0020: read 1 bytes at 0x0014: 0 (04)
[  202.573995] rmi4_i2c 1-0020: read 53 bytes at 0x0015: 0 (55 01 00 18 47 f9 37 37 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)
[  202.574033] input input5: rmi_2d_sensor_abs_report: obj[0]: type: 0x01 X: 393 Y: 1151 Z: 55 WX: 7 WY: 3
[  202.574049] input input5: rmi_2d_sensor_abs_report: obj[1]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0
[  202.574062] input input5: rmi_2d_sensor_abs_report: obj[2]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0
[  202.574075] input input5: rmi_2d_sensor_abs_report: obj[3]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0
[  202.574088] input input5: rmi_2d_sensor_abs_report: obj[4]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0
[  202.580378] rmi4_i2c 1-0020: read 1 bytes at 0x0014: 0 (04)
[  202.585783] rmi4_i2c 1-0020: read 53 bytes at 0x0015: 0 (55 01 00 18 47 f9 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)
[  202.585825] input input5: rmi_2d_sensor_abs_report: obj[0]: type: 0x01 X: 393 Y: 1151 Z: 0 WX: 0 WY: 0
[  202.585844] input input5: rmi_2d_sensor_abs_report: obj[1]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0
[  202.585861] input input5: rmi_2d_sensor_abs_report: obj[2]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0
[  202.585877] input input5: rmi_2d_sensor_abs_report: obj[3]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0
[  202.585894] input input5: rmi_2d_sensor_abs_report: obj[4]: type: 0x01 X: 0 Y: 0 Z: 0 WX: 0 WY: 0

```

Help needed to get my touchscreen working, thank you!

Minecrell, 2021-04-11:
@iyzsong Can you ping me in the [postmarketOS mainline chat](https://wiki.postmarketos.org/wiki/Matrix_and_IRC) for help? Thanks.

iyzsong, 2021-04-12:
Sure, I'll ping you there when I'm available for it, thank you!

iyzsong, 2021-04-18:
I made it somehow work by report `RMI_2D_OBJECT_NONE` when the pressure is 0:
```
diff --git a/drivers/input/rmi4/rmi_f01.c b/drivers/input/rmi4/rmi_f01.c
index d7603c50f864..4de9102c5145 100644
--- a/drivers/input/rmi4/rmi_f01.c
+++ b/drivers/input/rmi4/rmi_f01.c
@@ -564,7 +564,6 @@ static int rmi_f01_probe(struct rmi_function *fn)
                dev_err(&fn->dev,
                        "Device was reset during configuration process, status:
 %#02x!\n",
                        RMI_F01_STATUS_CODE(device_status));
-               return -EINVAL;
        }
 
        dev_set_drvdata(&fn->dev, f01);
diff --git a/drivers/input/rmi4/rmi_f11.c b/drivers/input/rmi4/rmi_f11.c
index 49ca9168685a..2b0c889b97ec 100644
--- a/drivers/input/rmi4/rmi_f11.c
+++ b/drivers/input/rmi4/rmi_f11.c
@@ -582,6 +582,10 @@ static void rmi_f11_abs_pos_process(struct f11_data *f11,
        obj->wx = pos_data[3] & 0x0f;
        obj->wy = pos_data[3] >> 4;
 
+       if (obj->z == 0) {
+         obj->type = RMI_2D_OBJECT_NONE;
+       }
+
        rmi_2d_sensor_abs_process(sensor, obj, n_finger);
 }
```

Close it for now..  Thank you!

stephan-gh, 2023-01-09:
Re-opening this since I think it is still unsolved. Ideally someone should work with the RMI4 maintainers upstream to get this handled in a sane way somehow.