msm8916-mainline/linux #229: samsung-j5x: Display starts to lighten up and may damage device

geraldog, 2022-03-21:
Hello, msm89160-mainline Linux people!

I'm from Brazil and experimenting with my Samsung J5 Metal J510MN phone. Patching lk2nd to accept fastboot by including Bootloader information that ties up to my Model to the right MUIC was the easy part. The hard part is that with PostmarketOS pmbootstrap I get a kernel that starts to light up the Samsung s6e8aa5x01-ams520kt01 panel with bizarre breakage. I'm pretty sure it will damage the display if left on.

What I do personally is I thank Samsung that there's an external battery to disconnect. So I've been playing for days and I found out that if I properly declare a driver for s6e8aa5x01-ams520kt01 and declare the panel in the DTS with mdss, dsi, dsi_phy nodes etc. plus the reset-gpio and some fixed-regulators for display power then I go further and get a black screen. It could have panic'ed the kernel for what I know, which is nothing, since I have no means presently to connect to serial UART nor USB Gadget Serial works.

I have now finally found @wonderfulShrineMaidenOfParadise fork of msm8916-mainline Linux with, from my perspective, closer to correct information on the DTS but I think there are some differences on the DSI DCS command-sequence of our panel drivers and thus original DTBs extracted with linux-mdss-dsi-panel-driver-generator tools.

Any feedback to get this working would be appreciated. I have some technical notes to share.

wonderfulShrineMaidenOfParadise, 2022-03-22:
Hi there,
Sadly the generated driver doesn't work well out of the box. I made some changes to make it working reasonably.
There are 2 changes, https://github.com/wonderfulShrineMaidenOfParadise/linux/commit/07d98e81feecbc0c8c6a57d0274fa58cfa9e1719 and https://github.com/wonderfulShrineMaidenOfParadise/linux/commit/336dc1034db86a437063456c2ed62407a651f87b

Before it fixed @marohinmark has managed to make `panel-simple` working. We may use it first. However panel-simple doesn't support sleeping either as far as I known. In another word, the panel will be stay powered on as well.

geraldog, 2022-03-22:
Oh, I see what you did now. Thank you very much for your work and for the prompt reply.

I'll try with simple-panel but there must be a gpio or something wrong on my DTS compared to your version of the j5x, because I get a black screen even though we're talking about the same panel.

The black screen is of course better than the exploding brightness that may damage the display permanently.

On a more technical note, on my decompiled downstream DTS I have the property qcom,mdss-dsi-force-clk-lane-hs listed for the panel. Shouldn't we disable the MIPI_DSI_MODE_LPM of the driver if we find such property?

stephan-gh, 2022-03-22:
> On a more technical note, on my decompiled downstream DTS I have the property qcom,mdss-dsi-force-clk-lane-hs listed for the panel. Shouldn't we disable the MIPI_DSI_MODE_LPM of the driver if we find such property?

qcom,mdss-dsi-force-clk-lane-hs is not related to MIPI_DSI_MODE_LPM, it's related to MIPI_DSI_CLOCK_NON_CONTINUOUS (see https://github.com/msm8916-mainline/linux-mdss-dsi-panel-driver-generator/blob/7cb9fb2de0fbdff6342295aab65e55dd38f4a155/panel.py#L248-L250).

geraldog, 2022-03-23:
Thanks for the quick reply, Stephan. If such was the case, I was misled by the [freedreno DSI Panel Driver Porting Guide](https://gitlab.freedesktop.org/freedreno/freedreno/-/wikis/DSI-Panel-Driver-Porting)

> If panel uses both high-speed and low-power mode, set MIPI_DSI_CLOCK_NON_CONTINUOUS.  Check the -command-state fields for dsi_lp_mode and dsi_hs_mode.  If you see both, then set MIPI_DSI_CLOCK_NON_CONTINUOUS since we are switching dynamically between different speeds. There's a catch though - another field (dsi-force-clk-lane-hs) exists, which indicates that only high-speed mode is used. The issue is, some downstream kernels may use both of these fields, and even set them different to one another. In that case, dsi-force-clk-lane-hs takes precedence, and you can safely ignore the -command-state fields.

It literally tells us to ignore the -command-state fields, and that only high-speed mode should be used.

geraldog, 2022-03-23:
> In any case, testing with and without MIPI_DSI_MODE_LPM is obviously a simple test you can do. At the end guidelines are just one thing - you want to get the panel working so any promising change could be a big step forward. :)

Oh, I tried and I tried, heh, apparently I need to try harder.

It doesn't bug me that @wonderfulShrineMaidenOfParadise j5x-panel branch got it working in slightly different model of J510, I'm actually glad I finally found the j5x-panel branch. Because it tied up the few missing holes in my DTS and driver. It doesn't work out of the box for me, but at least it gives a harmless black screen instead of countdown to permanent damage of display.

One more thing folks, is arch/arm64/boot/dts/qcom/msm8916-pins.dtsi gpio115, precisely what enables vdd-supply for the display in question, is assigned to primary Mi2S in ext-pri-tlmm-lines node of msmgpio. And to complicate things even more, gpio112, which in my downstream DTS stands for the display ESD detection pins gpio, is assigned to secondary Mi2S in ext-sec-tlmm-lines node of msmgpio

Mis012, 2022-03-23:
there are magic device tree statements which can be used to delete nodes or properties in included dtsi files, so I guess you could use that if your device indeed does things differently

as for display being black, that sounds suspiciously similar to what I was getting when towards the end of my journey in making a panel driver by hand - the last piece of the puzzle was that I had the polarity of the reset pin wrong (so at the end of sammy's magic on-off-on sequence that I had left in just in case, it was actually off-on-__off__)

I'm sure there are other possible reasons for the panel to stubbornly stay black, but I guess it's worth a try

wonderfulShrineMaidenOfParadise, 2022-04-23:
Sadly there is no better solution found to fix clock issue in generated driver e1ee4c33449e9772d370096c52abb5c917ca38d3. We are going to use `panel-simple` for now.
#237 should be able to enable panel with `panel-simple` driver. But it's still not possible to turn it off at the moment.

geraldog, 2022-05-03:
No point in leaving this open, then. I will give #237 a try and comment there. Thanks @wonderfulShrineMaidenOfParadise !

closed 2022-05-03 by geraldog