msm8916-mainline/linux #268: wt88047 problems : waking the screen.

exkc, 2022-12-21:
https://gitlab.com/postmarketOS/pmaports/-/issues/1080

panel-r61308 cant wake up after enter the sleep mode.

iamdavidcz, 2023-02-22:
@exkc @stephan-gh

The same issue happened to me recently on BQ Aquaris X5 (`paella`, `longcheer-l8910`) with `longcheer-yushun-nt35520` panel. I had a change to investigate it deeper and here's my two cents on this:

Apparently, `dsi_mgr_bridge_power_on(bridge)` is **not** called on subsequent times when the screen (panel) is woken up.
Consequently, the `power_on` flag on `msm_dsi_host` struct is `false` causing `dsi_host_transfer` to [return `-EINVAL`](https://github.com/msm8916-mainline/linux/blob/msm8916/6.2-rc5/drivers/gpu/drm/msm/dsi/dsi_host.c#L1641-L1642) (-22):
```
[  400.315351] panel-yushun-nt35520 1a98000.dsi.0: Failed to initialize panel: -22
[  400.315413] dsi_mgr_bridge_pre_enable: prepare panel 0 failed, -22
```
The following traceback is a consequence of this issue:
```
[  400.368926] msm_mdp 1a01000.mdp: vblank time out, crtc=0
[  410.455310] msm_mdp 1a01000.mdp: [drm] *ERROR* flip_done timed out
[  410.455351] msm_mdp 1a01000.mdp: [drm] *ERROR* [CRTC:57:crtc-0] commit wait timed out
[  420.481926] msm_mdp 1a01000.mdp: [drm] *ERROR* flip_done timed out
[  420.481965] msm_mdp 1a01000.mdp: [drm] *ERROR* [CONNECTOR:32:DSI-1] commit wait timed out
[  430.508563] msm_mdp 1a01000.mdp: [drm] *ERROR* flip_done timed out
[  430.508603] msm_mdp 1a01000.mdp: [drm] *ERROR* [PLANE:33:plane-0] commit wait timed out
[  430.508860] ------------[ cut here ]------------
[  430.508875] WARNING: CPU: 2 PID: 46 at drivers/gpu/drm/msm/disp/mdp5/mdp5_crtc.c:808 mdp5_crtc_atomic_flush+0x1a8/0x1c0
[  430.508962] Modules linked in: nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink bnep ipv6 ak8975 bmi160_i2c bmi160_core pm8xxx_vibrator snd_soc_apq8016_sbc snd_soc_qcom_common snd_soc_aw8738
[  430.509230] CPU: 2 PID: 46 Comm: kworker/2:1 Not tainted 6.1.12-msm8916 #2
[  430.509269] Hardware name: BQ Aquaris X5 (Longcheer L8910) (DT)
[  430.509290] Workqueue: events console_callback
[  430.509335] pstate: 20000005 (nzCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  430.509376] pc : mdp5_crtc_atomic_flush+0x1a8/0x1c0
[  430.509427] lr : drm_atomic_helper_commit_planes+0x150/0x200
[  430.509483] sp : ffff8000093ab760
[  430.509498] x29: ffff8000093ab760 x28: 0000000000000001 x27: ffff000002dedac8
[  430.509558] x26: 0000000000000000 x25: 0000000000000000 x24: 0000000000000020
[  430.509614] x23: 0000000000000070 x22: 0000000000000038 x21: ffff00000235cc00
[  430.509671] x20: ffff000002ded800 x19: ffff000002c20800 x18: 00000000fffffffd
[  430.509729] x17: 0048006f003e0528 x16: 0518051405280500 x15: 05000000038e0336
[  430.509787] x14: 032a038e02d002d0 x13: 0000000000000000 x12: 0000000000000003
[  430.509843] x11: 0000000000000000 x10: 0000000002d00000 x9 : 0000000000000000
[  430.509899] x8 : 00000000000002d0 x7 : 0000000000000001 x6 : 0000000000000000
[  430.509953] x5 : 0000000000000000 x4 : ffff000004a74900 x3 : ffff8000085c25d0
[  430.510009] x2 : ffff000002ded800 x1 : ffff000004a74280 x0 : ffff0000042eda00
[  430.510068] Call trace:
[  430.510081]  mdp5_crtc_atomic_flush+0x1a8/0x1c0
[  430.510134]  drm_atomic_helper_commit_planes+0x150/0x200
[  430.510186]  msm_atomic_commit_tail+0x190/0x4f0
[  430.510227]  commit_tail+0xa0/0x16c
[  430.510277]  drm_atomic_helper_commit+0x144/0x170
[  430.510329]  drm_atomic_commit+0x88/0x114
[  430.510373]  drm_client_modeset_commit_atomic+0x208/0x250
[  430.510410]  drm_client_modeset_commit_locked+0x58/0x18c
[  430.510445]  drm_fb_helper_pan_display+0xb8/0x1d0
[  430.510496]  fb_pan_display+0x94/0x110
[  430.510540]  bit_update_start+0x1c/0x40
[  430.510572]  fbcon_switch+0x3d0/0x520
[  430.510623]  redraw_screen+0x140/0x240
[  430.510675]  fbcon_blank+0x1f0/0x20c
[  430.510726]  do_unblank_screen+0x8c/0x140
[  430.510778]  poke_blanked_console+0xc8/0xd4
[  430.510812]  console_callback+0x110/0x150
[  430.510846]  process_one_work+0x1d0/0x320
[  430.510895]  worker_thread+0x144/0x434
[  430.510941]  kthread+0x108/0x110
[  430.510981]  ret_from_fork+0x10/0x20
[  430.511022] ---[ end trace 0000000000000000 ]---
```

The possible solution for this is to explicitly call `dsi_mgr_bridge_power_on(bridge)` before `drm_panel_prepare(panel)` is called:
```
diff --git a/drivers/gpu/drm/msm/dsi/dsi_manager.c b/drivers/gpu/drm/msm/dsi/dsi_manager.c
index 0b516a0..a24504d 100644
--- a/drivers/gpu/drm/msm/dsi/dsi_manager.c
+++ b/drivers/gpu/drm/msm/dsi/dsi_manager.c
@@ -25,6 +25,7 @@ struct msm_dsi_manager {
 
        bool is_bonded_dsi;
        bool is_sync_needed;
+       bool is_power_needed;
        int master_dsi_link_id;
 };
 
@@ -32,6 +33,7 @@ static struct msm_dsi_manager msm_dsim_glb;
 
 #define IS_BONDED_DSI()                (msm_dsim_glb.is_bonded_dsi)
 #define IS_SYNC_NEEDED()       (msm_dsim_glb.is_sync_needed)
+#define IS_POWER_NEEDED()      (msm_dsim_glb.is_power_needed)
 #define IS_MASTER_DSI_LINK(id) (msm_dsim_glb.master_dsi_link_id == id)
 
 #ifdef CONFIG_OF
@@ -412,8 +414,10 @@ static void dsi_mgr_bridge_pre_enable(struct drm_bridge *bridge)
        if (is_bonded_dsi && !IS_MASTER_DSI_LINK(id))
                return;
 
-       if (!dsi_mgr_power_on_early(bridge))
+       if (IS_POWER_NEEDED())
                dsi_mgr_bridge_power_on(bridge);
+       else
+               msm_dsim_glb.is_power_needed = true;
 
        /* Always call panel functions once, because even for dual panels,
         * there is only one drm_panel instance.
```

@stephan-gh Please test this patch and let me know the result :)

exkc, 2023-03-08:
@iamdavidcz Thank! https://gitlab.com/postmarketOS/pmaports/-/issues/1570

iamdavidcz, 2023-03-08:
@exkc
Nps, I'm using this patch for about 2 weeks now on my `paella` device and it works fine for me, but it would be nice if pmOS guys test it too :)

exkc, 2023-03-08:
@iamdavidcz 

I tested it and 
[image]

exkc, 2023-03-08:
Hua it is odd sddm (it probably running on xorg) seem to work fine but after i loged in to xfce this happent

exkc, 2023-03-08:
Ah i see why routing the screen in xorg cusred this issue

exkc, 2023-03-08:
Ah i see why log in to xfce4 caused this issue bc i configed my xfce4 to rotate my screen.

rotate the screen caused  that issue.

exkc, 2023-03-08:
the patch also caused this when i turn off the screen on kde mobile (wayland )
[image]
(and the back light on but nothing being display)

exkc, 2023-03-08:
This patch fixed the issue and caused two issue NICE.

iamdavidcz, 2023-03-08:
@exkc
This could be panel specific, on `paella` panel initialization works fine after wake up.

exkc, 2023-03-09:
> This could be panel specific, on paella panel initialization works fine after wake up.

Have tested it on wayland and rotate the screen on xorg/wayland ?

Jakko3, 2023-05-07:
I would guess the commit https://github.com/msm8916-mainline/linux/commit/79bfde63e796a6142926323c22acfe681ed1e3dc Revert "drm/msm/dsi: switch to DRM_PANEL_BRIDGE" is somehow involved. (The commit id might change on new releases, the current one can be found in drivers/gpu/drm/msm -> button "History".)

I had a try just reverting it (reverting the revert). That didn't solve anything.

stephan-gh, 2023-05-12:
> I would guess the commit https://github.com/msm8916-mainline/linux/commit/79bfde63e796a6142926323c22acfe681ed1e3dc Revert "drm/msm/dsi: switch to DRM_PANEL_BRIDGE" is somehow involved. (The commit [...]

AFAIK the issue was already present before we reverted that one. And at the moment this revert is still necessary (again AFAIK) to get display disable sequences working properly. Otherwise you get an error there as well.

Ro-Den, 2023-05-20:
Don't know if it's related: https://github.com/antony-jr/postmarketos-asus-z00l/issues/3

Should I open a separate issue here? Or my issue is device specific, isn't it?

Jakko3, 2023-05-20:
On my devices in X11 desktop environments, after the display went to sleep, at wake-up in command dmesg (or dmesg -W to follow via ssh) I get:

bq-paella:
[ 4508.397864] panel-yushun-nt35520 1a98000.dsi.0: transmit data failed: -22
[ 4508.397924] panel-yushun-nt35520 1a98000.dsi.0: Failed to initialize panel: -22
[ 4508.397976] dsi_mgr_bridge_pre_enable: prepare panel 0 failed, -22
[ 4508.451210] msm_mdp 1a01000.display-controller: vblank time out, crtc=0
[ 4518.614492] msm_mdp 1a01000.display-controller: [drm] *ERROR* flip_done timed out
[ 4518.614550] msm_mdp 1a01000.display-controller: [drm] *ERROR* [CRTC:57:crtc-0] commit wait timed out
[ 4528.641148] msm_mdp 1a01000.display-controller: [drm] *ERROR* flip_done timed out
[ 4528.641206] msm_mdp 1a01000.display-controller: [drm] *ERROR* [CONNECTOR:32:DSI-1] commit wait timed out

samsung-serranove:
[  359.679279] panel-s6e88a0-ams427ap24 1a98000.dsi.0: sending command 0xf0 failed: -22
[  359.679300] panel-s6e88a0-ams427ap24 1a98000.dsi.0: Failed to initialize panel: -22
[  359.679524] dsi_mgr_bridge_pre_enable: prepare panel 0 failed, -22
[  359.733002] msm_mdp 1a01000.display-controller: vblank time out, crtc=0
[  369.903438] msm_mdp 1a01000.display-controller: [drm] *ERROR* flip_done timed out
[  369.903497] msm_mdp 1a01000.display-controller: [drm] *ERROR* [CRTC:57:crtc-0] commit wait timed out
[  379.929913] msm_mdp 1a01000.display-controller: [drm] *ERROR* flip_done timed out
[  379.929967] msm_mdp 1a01000.display-controller: [drm] *ERROR* [CONNECTOR:32:DSI-1] commit wait timed out

When reverting commit 79bfde6, back to mainline, there is the message:
[ 1094.091203] panel-s6e8aa5x01-ams497hy01 1a98000.dsi.0: Failed to set display off: -22
[ 1094.091267] panel-s6e8aa5x01-ams497hy01 1a98000.dsi.0: Failed to un-initialize panel: -22

Btw. I currently work around this issue by avoiding the panel going to sleep. When locking the screen, it turns black and can be awakened again. Unfortunately, with this workaround the backlight stays on, which is not a good solution.
- In Xfce4 -> Settings -> Power Manager -> Tab "Display" -> uncheck "Display Power Management"
- In MATE -> System -> Preferences -> Hardware -> Power Management -> tabs "On AC Power" and "On Battery Power" set "put diesplay to sleep..." to value "Never", if not selectable by touchscreen the click once, call the onboard keyboard and use the arrow up/down keys).

Desktop environments using Wayland like Phosh are not affected.

----

I see asus-z00l is a MSM8916 Qualcomm Snapdragon 410 as well, so possibly in X11 desktop environments you would have the same issue. However, in the issue described here the display and backlight works fine right after boot – the display just refused to wake up once it fell asleep. In your issue it's the backlight itself not turning on for whatever reason.

On your question to open a separate issue here: I'm not sure if your issue is kernel-related, driver or devicetree (belongs here) or user-interface-related (issue to postmarketos/pmaports). Likely it's kernel-related and belongs here. In case you can easily switch sd-card with another user interface, I would propose you first test if the issue persists on another user interface as well. If it's also existent in other user interfaces or if you can't easily switch to another user interface, then I propose you set up a new issue here.

In the meantime you probably can work around your issue by an udev rule.
- command `udevadm info --attribute-walk /sys/class/backlight/1a98000.dsi.0`
  - look for SUBSYSTEM and KERNEL
- `sudo vi /etc/udev/rules.d/20_asus-z00l_backlight.rules` (new file)
  - type something like `ACTION=="add", SUBSYSTEM=="backlight", KERNEL=="...", ATTR{brightness}="100"` where you need to change the name of KERNEL, safe by Esc and typing `:wq`

exkc, 2023-05-25:
Ah btw i switched to samsung a5 for a while this issue happented as well

> In case you can easily switch sd-card with another user interface, 

Ah , BTW i use gnome kde (big screen/desptop/mobile) xfce4 icewm on samsung a5

vknecht, 2023-05-30:
Been having that sort of problems for quite some time with AOSP, worked-around with https://github.com/vknecht/msm8916-mainline-linux/commit/ceba76ec3e5fbc02cf0a49ad02b4bebf133848ff

There was a discussion
https://lore.kernel.org/linux-arm-msm/CAA8EJpq_9iC1rkiZVom28Kv_B3QLd4pBgFObxBfSpJ+Xh=Mp1g@mail.gmail.com/
where it was suggested to split .prepare() / .unprepare() by using new .enable() /.disable() callbacks introduced by
https://lore.kernel.org/dri-devel/20221205173328.1395350-1-dave.stevenson@raspberrypi.com/

Didn't get to the bottom of it, maybe that should be done in panel generator
if that works, and maybe could help with not having to do this revert https://github.com/msm8916-mainline/linux/commit/7f74040bd869f69da0282720bccee845f3224da6

Jakko3, 2023-06-04:
Nice hints, that's a notable step foreward!

I want to share my state of research, although it's not finished. Still it might help the discussion (I hope it helps and does not mislead).

On samsung-serranove the issue started between `v5.17-msm8916` and `v5.18-msm8916`.

I wanted to do a "git bisect", which would need to rebase the mainline (torvalds) `v5.18` commits on top of `v5.17-msm8916`. I failed to do this rebase cleanly, therefore didn't even get to the "git bisect".

Instead started to poke around in `v5.18-msm8916`. A candidate causing the issue is commit 7d8e9a90509f "drm/msm/dsi: move DSI host powerup to modeset time", which is also mentioned in vknecht's hack. Reverting that commit in `v5.18-msm8916` makes my display to wake up as it should.

However, reverting that commit is not a sustainable solution. I tried to understand what the commit acctually does. The change is not easy to understand to me. Actually it makes of function `dsi_mgr_bridge_pre_enable()` in `v5.17-msm8916`...
- https://github.com/msm8916-mainline/linux/blob/v5.17-msm8916/drivers/gpu/drm/msm/dsi/dsi_manager.c#L339

... two functions `dsi_mgr_bridge_power_on()` and `dsi_mgr_bridge_pre_enable()` in `v5.18-msm8916`, which are slighty different from the original function in `v5.17-msm8916`.
- https://github.com/msm8916-mainline/linux/blob/v5.18-msm8916/drivers/gpu/drm/msm/dsi/dsi_manager.c#L318
- https://github.com/msm8916-mainline/linux/blob/v5.18-msm8916/drivers/gpu/drm/msm/dsi/dsi_manager.c#L374

I don't yet have a good understanding what it means. A bit simplified I would say that the content of "bridge pre-enable" in `v5.17-msm8916` was split into "bridge power on" and "bridge pre-enable" in `v5.18-msm8916`. Now the "bridge pre-enable" alone contains less things than before, because some stuff moved into "bridge power on".

Setting some additional debug flags, I could see the order of functions called in `v5.18-msm8916`:
- at boot:
  - bridge mode set
  - bridge power on
  - bridge pre-enable
  - panel prepare
  - panel on
  - bridge enable
  - (panel enable) *# if implemented*
- going to sleep:
  - bridge disable
  - (panel disable) *# if implemented*
  - bridge post-disable
  - panel unprepare
  - panel off
- waking up:
  - bridge pre-enable
  - panel prepare
  - panel on
  - bridge enable
  - (panel enable) *# if implemented*

So the new function "bridge power on" introdced in `v5.18-msm8916` is called at boot but not at wake-up. And the function "bridge mode set", which would call "bridge power on", is also called just at boot but not at wake-up. As some of the stuff of "bridge pre-enable" moved into "bridge power on", they are not called anymore on wake-up in `v5.18-msm8916`.

Notably there is the function `msm_dsi_host_power_on()` that moved from `dsi_mgr_bridge_pre_enable()` to `dsi_mgr_bridge_power_on()` in `v5.18-msm8916`.

Interstingly at the end of function `dsi_mgr_bridge_post_disable()` there is `msm_dsi_host_power_off()`. So at "bridge post-disable" the dsi host gets powered off. But when waking up that host doesn't get powered on because that would happen in "bridge power on" which is not called at wake-up.

That's basically where I currently am with the research.

I just had a short glimpse into the current kernel (the above was just about `v5.18-msm8916`). In `msm8916/6.4-rc4` I see within "bridge pre-enable" there is now a call of "bridge power on" (that's what we need) but under the condition that `dsi_mgr_power_on_early()` isn't valid.
- https://github.com/msm8916-mainline/linux/blob/msm8916/6.4-rc4/drivers/gpu/drm/msm/dsi/dsi_manager.c#L415-L416

Using the "blame" button to see where this condition comes from shows that it was introduced by commit ec7981e6c614 "drm/msm/dsi: don't powerup at modeset time for parade-ps8640". The commit diff also shows that the "brige power on" should actually happen in "bridge mode set" function.

This closes the circle to vknecht's hack to not power on early. I'm not sure all the text here added something at all ;) Nonetheless I think we're getting closer.

As a next step it would need a better understanding why the `dsi_mgr_bridge_power_on()` is in function `dsi_mgr_bridge_mode_set()`. Looks like this function gets called at boot but not at panel wake-up (so far I just tested on `v5.18-msm8916`).

vknecht, 2023-06-06:
The latest few commits from Doug at
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/log/drivers/gpu/drm/msm/dsi/dsi_manager.c
have some more description of the situation and fix,
especially since the bridge workaround I piggy-backed on is now gone

Jakko3, 2023-06-06:
That's great news! This commit solves our issue: 
- https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/gpu/drm/msm/dsi/dsi_manager.c?id=9e15123eca7942caa8a3e1f58ec0df7d088df149

Well, solves one of two issues, to be more precise. For the other one, however, I already have a rough proposal. Maybe it needs a bit fine-tuning.

I looked into the current kernel `msm8916/6.4-rc4`, device samsung-serranove, UI xfce4. The first thing I did was reverting msm8916-patch https://github.com/msm8916-mainline/linux/commit/9de91138caeb296f773b0f3fa846073a5bbac3bd (the hash number will be different on other msm8916 kernel versions) – to be closer to mainline. Secondly I added additional debug markers to see the path we're taking.

Interesingly, in `msm8916/6.4-rc4` the order is different than in `v5.18-msm8916`.

- at boot:
  - bridge mode set
  - bridge power on
  - panel prepare
  - panel on
  - bridge pre-enable
  - (panel enable) *# if implemented*

- going to sleep:
  - (panel disable) *# if implemented*
  - bridge post-disable
  - panel unprepare
  - panel off

- waking up:
  - panel prepare
  - panel on
  - bridge pre-enable
  - (panel enable) *# if implemented*

There is no "bridge enable" and "bridge disable" anymore. And there are other differences in the order, in more detail, the above list is not fine-grained enough. Specifically the location where "panel unprepare" gets called had changed. In `v5.18-msm8916` it was in the middle of `dsi_mgr_bridge_post_disable()` [1]. In `msm8916/6.4-rc4` with https://github.com/msm8916-mainline/linux/commit/9de91138caeb296f773b0f3fa846073a5bbac3bd reverted there is no call to unprepare the panel wihtin `dsi_mgr_bridge_post_disable()` [2], so it was called after "bridge post-disable" had finished.
- [1] https://github.com/msm8916-mainline/linux/blob/v5.18-msm8916/drivers/gpu/drm/msm/dsi/dsi_manager.c#L525-L530
- [2] https://github.com/torvalds/linux/blob/v6.4-rc4/drivers/gpu/drm/msm/dsi/dsi_manager.c#L369-L421

And there is a new 2nd issue at display goint to sleep with "Failed to set display off: -22". This is caused by commit https://github.com/msm8916-mainline/linux/commit/007ac0262b0dbb5523e99f03c6f3f44e09b15f66 and temporarily solved in msm8916-mainline by the already mentioned commit https://github.com/msm8916-mainline/linux/commit/9de91138caeb296f773b0f3fa846073a5bbac3bd, which reverts the other one.

So currently we have two issues: One caused by https://github.com/msm8916-mainline/linux/commit/7d8e9a90509f1bd1d193a0c93cb8d1dbad9049fb in `v5.18-msm8916` failing panel wake-up and one by https://github.com/msm8916-mainline/linux/commit/7d8e9a90509f1bd1d193a0c93cb8d1dbad9049fb that came up in `v6.1-msm8916` failing panel sleep. This second issue seems to be caused by the changed order of calling "panel unprepare". In the second half of "bridge post-disable" several things are turned off, notably `msm_dsi_host_power_off()`. So in `v5.18-msm8916` "panel unprepare" was before `msm_dsi_host_power_off()`, in `msm8916/6.4-rc4` it's after – and failing because the dsi host isn't available anymore.

The solution of this scond issue seems to be to use "panel disable" callback, this is called before "bridge post-disable". I'm not yet fully sure if just moving "panel off" into "panel disable" is a clean enough solution or whether there are more elegant or fine-grained ways. Anyway, applying this change solves the second issue for me on device samsung-serranove.

Additionally, I needed to move "panel on" into "panel enable". Because the new commit mentioned by vknecht moved the `dsi_mgr_bridge_power_on()` back from "bridge mode set" to "bridge pre-enable". This is called after "panel prepare" – making "panel on" fail because `dsi_mgr_bridge_power_on()` wasn't called yet. Moving "panel on" into "panel enable", which is called after "bridge pre-enable", solves this. Again I'm not fully sure if just moving the "panel on" entirely is clean enough or whether it needs a more fine-grained handling. In noticed some display flickering during panel on/off, so possibly this handling is not smooth enough.

For those who want to test:
- `git clone https://github.com/msm8916-mainline/linux.git`
- `cd linux`
- `git checkout msm8916/6.4-rc4`
- `git checkout -b display_test`
- `git revert 9de91138caeb` # this reverts msm8916 temporary fix "Revert "drm/msm/dsi: switch to DRM_PANEL_BRIDGE""
- `git remote add linux-next https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git`
- `git fetch linux-next`
- `git cherry-pick 9e15123eca79` # that's the commit "drm/msm/dsi: Stop unconditionally powering up DSI hosts at modeset", fixing the first issue
- `git cherry-pick d8dd416cb420` # another commit "drm/msm/dsi: More properly handle errors in regards to dsi_mgr_bridge_power_on()", shouldn't make a difference in our case
- edit your panel to fix the second issue, the following diff is on the panel of samsung-serranove:
```diff
diff --git a/drivers/gpu/drm/panel/msm8916-generated/panel-samsung-s6e88a0-ams427ap24.c b/drivers/gpu/drm/panel/msm8916-generated/panel-samsung-s6e88a0-ams427ap24.c
index 46bd1eb67d32..42c3d5f44d22 100644
--- a/drivers/gpu/drm/panel/msm8916-generated/panel-samsung-s6e88a0-ams427ap24.c
+++ b/drivers/gpu/drm/panel/msm8916-generated/panel-samsung-s6e88a0-ams427ap24.c
@@ -113,6 +113,24 @@ static int s6e88a0_ams427ap24_off(struct s6e88a0_ams427ap24 *ctx)
 	return 0;
 }
 
+static int s6e88a0_ams427ap24_enable(struct drm_panel *panel)
+{
[... cut ...]
 static const struct drm_display_mode s6e88a0_ams427ap24_mode = {
 	.clock = (540 + 94 + 4 + 18) * (960 + 12 + 1 + 3) * 60 / 1000,
 	.hdisplay = 540,
@@ -196,8 +213,10 @@ static int s6e88a0_ams427ap24_get_modes(struct drm_panel *panel,
 }
 
 static const struct drm_panel_funcs s6e88a0_ams427ap24_panel_funcs = {
+	.enable = s6e88a0_ams427ap24_enable,
 	.prepare = s6e88a0_ams427ap24_prepare,
 	.unprepare = s6e88a0_ams427ap24_unprepare,
+	.disable = s6e88a0_ams427ap24_disable,
 	.get_modes = s6e88a0_ams427ap24_get_modes,
 };
```

stephan-gh, 2023-06-06:
> The solution of this scond issue seems to be to use "panel disable" callback, this is called before "bridge post-disable". I'm not yet fully sure if just moving "panel off" into "panel disable" is a clean enough solution or whether there are more elegant or fine-grained ways. 

AFAIK most of the panel drivers upstream still have it the way it's currently in all the "msm8916-generated" panel drivers, most of the DSI commands are sent in prepare() and unprepare(), not in enable()/disable(). I've seen some related discussions over time but not actual conclusions. The [original regression report](https://lore.kernel.org/linux-arm-msm/1892d0af-8aa4-dc03-887a-e44bb623b836@linaro.org/) also didn't really end up anywhere.

Note that the panel subsystem is independent from the display controller drivers (MDSS here). Just because the Qualcomm MDSS driver needs those to be in enable()/disable() doesn't mean the same applies to all other display controller drivers as well. A panel driver is supposed to be usable with different display controllers.

This is why I just reverted the problematic commit and kept the panel drivers as-is.

I would like to see a proper conclusion upstream, which would involve either fixing the MDSS driver or adjusting all panel drivers upstream. Perhaps someone could investigate this further and/or get a discussion about this started again upstream.

Jakko3, 2023-06-08:
Ok, when linux-next commit 9e15123eca79 reaches the kernel, the reported issue should be solved.

The other topic is how to get rid of the temporary fix 9de91138caeb of issue 2.

stephan-gh, 2023-06-08:
Can someone check if the `pre_enable_prev_first` flag introduced in that commit makes any difference for issue 2? Seems like it was set for some Samsung panels in https://github.com/torvalds/linux/commit/4fb912e5e190. Make sure to use it together with https://github.com/msm8916-mainline/linux/commit/9e15123eca7942caa8a3e1f58ec0df7d088df149 and do `git revert 9de91138caeb` to actually get the bridge behavior.

I don't quite fully understand what it does yet, is it only related to pre_enable() (which is working fine) or also the disable sequence?

Jakko3, 2023-06-08:
Starting from `msm8916/6.4-rc4`, I reverted 9de91138caeb, cherry-picked 9e15123eca79 and d8dd416cb420, added additional debug markers to follow the path. I get the following order. Missing numbers in the path are due to if-clauses either being skipped or taken. Additionally, I added empty enable/disable functions to the panel, just to track their appearance along the path.

The panel fails at boot because "panel on" gets called before "bridge pre-enable", thus before `dsi_mgr_bridge_power_on()`. Same thing on wake-up.

This is samsung-serranove on Xfce4 (X11). Phosh might be different (Wayland).

At boot:
```
[    1.337843] msm_dsi_manager_set_split_display: set split display 1
[    1.347565] msm_dsi_manager_set_split_display: set split display 3
[    1.347588] msm_dsi_manager_set_split_display: set split display 4
[    1.377288] dsi_mgr_bridge_mode_valid: bridge mode valid 1
[    1.387511] dsi_mgr_bridge_mode_valid: bridge mode valid 2
[    1.392803] dsi_mgr_bridge_mode_valid: bridge mode valid 4
[    1.398273] dsi_mgr_bridge_mode_valid: bridge mode valid 6
[    1.408328] dsi_mgr_bridge_mode_valid: bridge mode valid 1
[    1.408337] dsi_mgr_bridge_mode_valid: bridge mode valid 2
[    1.408344] dsi_mgr_bridge_mode_valid: bridge mode valid 4
[    1.408348] dsi_mgr_bridge_mode_valid: bridge mode valid 6
[    1.408430] dsi_mgr_bridge_mode_set: bridge mode set 1
[    1.408434] dsi_mgr_bridge_mode_set: bridge mode set 2
[    1.408437] dsi_mgr_bridge_mode_set: bridge mode set 4
[    1.408440] dsi_mgr_bridge_mode_set: bridge mode set 6
[    1.408477] s6e88a0_ams427ap24_prepare: panel prepare 1
[    1.408481] s6e88a0_ams427ap24_prepare: panel prepare 2
[    1.408942] s6e88a0_ams427ap24_prepare: panel prepare 3
[    1.436168] s6e88a0_ams427ap24_prepare: panel prepare 4
[    1.436177] s6e88a0_ams427ap24_on: panel on 1
[    1.436185] panel-s6e88a0-ams427ap24 1a98000.dsi.0: sending command 0xf0 failed: -22
[    1.436194] panel-s6e88a0-ams427ap24 1a98000.dsi.0: Failed to initialize panel: -22
[    1.436493] dsi_mgr_bridge_pre_enable: bridge pre-enable 1
[    1.436498] dsi_mgr_bridge_pre_enable: bridge pre-enable 2
[    1.436501] dsi_mgr_bridge_pre_enable: bridge pre-enable 3
[    1.436504] dsi_mgr_bridge_power_on: bridge power on 1
[    1.436506] dsi_mgr_bridge_power_on: bridge power on 2
[    1.436509] dsi_mgr_phy_enable: phy enable 1
[    1.436512] dsi_mgr_phy_enable: phy enable 5
[    1.438070] dsi_mgr_phy_enable: phy enable 6
[    1.438074] dsi_mgr_phy_enable: phy enable 7
[    1.438078] dsi_mgr_bridge_power_on: bridge power on 3
[    1.465073] dsi_mgr_bridge_power_on: bridge power on 4
[    1.465080] dsi_mgr_bridge_power_on: bridge power on 5
[    1.465089] dsi_mgr_bridge_power_on: bridge power on 6
[    1.465092] dsi_mgr_bridge_pre_enable: bridge pre-enable 4
[    1.465096] dsi_mgr_bridge_pre_enable: bridge pre-enable 5
[    1.465100] dsi_mgr_bridge_pre_enable: bridge pre-enable 6
[    1.465125] s6e88a0_ams427ap24_enable: panel enable 1

[   20.204814] dsi_mgr_bridge_mode_valid: bridge mode valid 1
[   20.204829] dsi_mgr_bridge_mode_valid: bridge mode valid 2
[   20.204838] dsi_mgr_bridge_mode_valid: bridge mode valid 4
[   20.204845] dsi_mgr_bridge_mode_valid: bridge mode valid 6
[   20.206081] dsi_mgr_bridge_mode_valid: bridge mode valid 1
[   20.206096] dsi_mgr_bridge_mode_valid: bridge mode valid 2
[   20.206105] dsi_mgr_bridge_mode_valid: bridge mode valid 4
[   20.206111] dsi_mgr_bridge_mode_valid: bridge mode valid 6
[   25.381232] dsi_mgr_bridge_mode_valid: bridge mode valid 1
[   25.381246] dsi_mgr_bridge_mode_valid: bridge mode valid 2
[   25.381254] dsi_mgr_bridge_mode_valid: bridge mode valid 4
[   25.381259] dsi_mgr_bridge_mode_valid: bridge mode valid 6
```

Going to sleep:
```
[   89.621788] s6e88a0_ams427ap24_disable: panel disable 1
[   89.638099] dsi_mgr_bridge_post_disable: bridge post-disable 1
[   89.638138] dsi_mgr_bridge_post_disable: bridge post-disable 2
[   89.638161] dsi_mgr_bridge_post_disable: bridge post-disable 3
[   89.664962] dsi_mgr_bridge_post_disable: bridge post-disable 4
[   89.664998] dsi_mgr_bridge_post_disable: bridge post-disable 7
[... 185 lines cut ...]
[  581.642777] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  581.719434] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  581.796095] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  581.872764] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  581.949592] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  582.026093] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  582.102764] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  582.132887] s6e88a0_ams427ap24_on: panel on 2
[  582.132887] s6e88a0_ams427ap24_on: panel on 2
[  582.206108] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  582.282768] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  582.359437] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  582.436176] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  582.512774] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[... cut ...]
[  582.819438] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  582.896092] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  582.972783] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  583.049430] msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out
[  583.052755] s6e88a0_ams427ap24_on: panel on 3
[  583.052787] s6e88a0_ams427ap24_prepare: panel prepare 5
[  583.052873] s6e88a0_ams427ap24_enable: panel enable 1
```

Jakko3, 2023-06-09:
Commit https://github.com/msm8916-mainline/linux/commit/5ea6b17027810ffbdb5bea7d0a2b1d312dd1021c introduced "panel->prepare_prev_first", commit https://github.com/msm8916-mainline/linux/commit/4fb912e5e19075874379cfcf074d90bd51ebf8ea introduced "bridge->pre_enable_prev_first".

The second one is supposed to act on "post_disable" as well:

> If a bridge sets @pre_enable_prev_first, then the @post_disable for that
> bridge will be called before the previous one to reverse the @pre_enable
> calling direction.

- https://github.com/torvalds/linux/blob/v6.4-rc4/drivers/gpu/drm/drm_bridge.c#L653-L655

It doesn't work in my case, for some reason.

stephan-gh, 2023-06-09:
The order doesn't really look great in both cases...

exkc, 2023-06-09:
> Phosh might be different (Wayland).

yes it should be different otherwise turn off the screnn is broken on both x11 n wayland

vknecht, 2023-06-13:
Caleb sent a patch for sdm845 panels :
https://lore.kernel.org/linux-arm-msm/20230613-panel-dsi-disable-v1-1-5e454e409fa8@linaro.org/T/#u

Jakko3, 2023-06-18:
It's still unclear why the panel option `ctx->panel.prepare_prev_first = true;` doesn't reverse the procedure of "post_disable" as it acutally should. At least on my device samsung-serranvoe. Therefore I added additional debug markers to `drivers/gpu/drm/drm_bridge.c`. It was quite some work and the result is rather hard to understand, at least to me.

I had initially intended to post the outcome here. But now I posted it to the thread that vknecht linked above. Actually I didn't want to disturb the discussion ongoing there. On the other hand, I hope some of the people in the thread have a deeper understanding of the decision-making within drm_bridge.c, so it's probably the right place to post it.
- https://lore.kernel.org/linux-arm-msm/20230613-panel-dsi-disable-v1-1-5e454e409fa8@linaro.org/T/#m9a8b8928b3657af2fd91baee78feeb1101da7b25

stephan-gh, 2023-08-01:
On the `wip/msm8916/6.5-rc4` branch I dropped the `drm/msm/dsi: switch to DRM_PANEL_BRIDGE` revert. Unfortunately, it still behaves exactly like @Jakko3 described. Without `prepare_prev_first` the display does not work at all, when setting it to `true` it works somewhat but there is spam in dmesg. And disabling the panel still doesn't work.

Looks like more investigation is needed :/

stephan-gh, 2023-08-02:
> It's still unclear why the panel option `ctx->panel.prepare_prev_first = true;` doesn't reverse the procedure of "post_disable" as it acutally should.

This part seems to be fixed by https://lore.kernel.org/dri-devel/20230328170752.1102347-1-jagan@amarulasolutions.com/

The other problem, related to the `msm_dsi 1a98000.dsi: [drm:dsi_cmds2buf_tx [msm]] *ERROR* wait for video done timed out` is much more involved. IMHO the problem is that the msm_dsi driver enables the video stream for video mode panels too early. The `pre_enable()` hook of the panel should be called with the DSI host powered up, but not actively streaming yet. This got broken in `drm/msm/dsi: switch to DRM_PANEL_BRIDGE`. The proper way to fix that is splitting the `pre_enable()`/`post_disable()` callbacks in the msm_dsi driver, so that "power up" and "start streaming" is split into two callbacks as before: I did this in https://github.com/msm8916-mainline/linux/commit/4ded90232c49bc7f25c9b8822c7fd60a73955109. With that (and the above post_disable order fix), panel initialization and deinitialization works fine.

However, this breaks the ordering between the "encoder" of MDP5 (the display controller) and the "bridge" of msm_dsi. For some reason `msm_dsi` must start streaming before the MDP5 part gets enabled. With my change, this order is swapped. I worked around this with a hack, but that part is far from being clean: https://github.com/msm8916-mainline/linux/commit/2dd625bf4885d153eecdc0b13ea976973f15f716 I don't know how to handle this cleanly yet.

For now I put these commits into `msm8916/6.5-rc4` as replacement for the revert. With these display seems to work just fine as before.

I don't quite see how this setup would make any difference for Xorg and other UIs but testing is still welcome of course, maybe we are lucky :)

stephan-gh, 2023-09-10:
I'll close this since it might be fixed in 6.5.2. Please comment here if this is still happening with the latest kernel release!

closed 2023-09-10 by stephan-gh

lm2lm2, 2024-09-25:
hello

feel a bit bad to report it, as it's proposed here : 
https://gitlab.com/postmarketOS/linux-postmarketos-archive/-/issues/26#note_898898757
on gprimeltexx, i still have several times per day the : 
[drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000

eg

> [20029.216104] wlan0: associated
> [20030.953680] wcn36xx: ERROR hal_enter_bmps response failed err=1
> [20030.953713] wcn36xx: ERROR Can not enter BMPS!
> [20043.285234] wcn36xx: ERROR hal_enter_bmps response failed err=1
> [20043.285286] wcn36xx: ERROR Can not enter BMPS!
> [20151.500031] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
> [23240.581273] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
> [23247.321639] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
> [23669.851997] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
> [23672.405304] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
> [23930.087662] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
> [25414.918569] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
> [25421.471876] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
> [25430.135143] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
> [27874.974428] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
> [33110.887392] [drm:mdp5_irq_error_handler [msm]] *ERROR* errors: 04000000
>

stephan-gh, 2024-09-26:
> [drm:mdp5_irq_error_handler [msm]] ERROR errors: 04000000

You can ignore those if the display is working without any problems.

stephan-gh, 2024-09-26:
No, the message is unrelated to the touchscreen. It's just about the displayed screen contents. That's probably something different...