msm8916-mainline/linux #282: samsung-a7: framebuffer doesn't refresh on 6.2-rc*

wonderfulShrineMaidenOfParadise, 2023-01-29:
## Steps to reproduce

1. Install 6.1. Framebuffer refreshes
2. Install 6.2-rc*. Framebuffer doesn't refresh
3. `Run kmscube` or refersh framebuffer manually
4. Install `msm-fb-refresher` and reboot. Framebuffer refreshes
5. Downgrade the kernel to 6.1 and remove `msm-fb-refresher` . Framebuffer still refreshes

Ref: https://wiki.postmarketos.org/wiki/Troubleshooting:display#Screen_does_not_refresh

stephan-gh, 2023-01-30:
Given that only a7 seems to be affected I'm guessing this is part of `#sad-display-command-mode-problems` :/

stephan-gh, 2023-02-20:
Hm, is only framebuffer affected or also proper UIs like Phosh? Those should not go through the framebuffer interface but through DRM.

wonderfulShrineMaidenOfParadise, 2023-02-25:
Only framebuffer is affected and Phosh works well.

wonderfulShrineMaidenOfParadise, 2024-03-26:
Cannot reproduce on 6.8. Closing.

closed 2024-03-26 by wonderfulShrineMaidenOfParadise