msm8916-mainline/linux #313: The pm8916-bms-vm interrupt triggers every 50s

aa889788, 2023-05-16:
the driver set the sample interval default to 100ms, and fifo count to 2, the irq should trigger every 200ms, but it triggers about 50s each
```bash
~ # dmesg | grep pm8916
[    0.686436] genirq: Flags mismatch irq 28. 00002003 (200f000.spmi:pmic@0:extcon@1300) vs. 00002003 (pm8916_lbc)
[   51.864523] pm8916_bms_vm_fifo_update_done_irq
[  105.370233] pm8916_bms_vm_fifo_update_done_irq
[  158.875939] pm8916_bms_vm_fifo_update_done_irq
[  212.382626] pm8916_bms_vm_fifo_update_done_irq
[  265.888328] pm8916_bms_vm_fifo_update_done_irq
[  319.395013] pm8916_bms_vm_fifo_update_done_irq
[  372.900722] pm8916_bms_vm_fifo_update_done_irq
[  426.406423] pm8916_bms_vm_fifo_update_done_irq
[  479.913109] pm8916_bms_vm_fifo_update_done_irq
[  533.419793] pm8916_bms_vm_fifo_update_done_irq
[  586.925501] pm8916_bms_vm_fifo_update_done_irq
[  640.432188] pm8916_bms_vm_fifo_update_done_irq
[  693.938874] pm8916_bms_vm_fifo_update_done_irq
[  747.445552] pm8916_bms_vm_fifo_update_done_irq
[  800.951261] pm8916_bms_vm_fifo_update_done_irq
[  854.456972] pm8916_bms_vm_fifo_update_done_irq
[  907.963650] pm8916_bms_vm_fifo_update_done_irq
[  961.469359] pm8916_bms_vm_fifo_update_done_irq
[ 1014.976042] pm8916_bms_vm_fifo_update_done_irq
[ 1068.482722] pm8916_bms_vm_fifo_update_done_irq
[ 1121.989405] pm8916_bms_vm_fifo_update_done_irq
[ 1175.496094] pm8916_bms_vm_fifo_update_done_irq
[ 1229.002771] pm8916_bms_vm_fifo_update_done_irq
[ 1282.508483] pm8916_bms_vm_fifo_update_done_irq
[ 1336.015164] pm8916_bms_vm_fifo_update_done_irq
[ 1389.520873] pm8916_bms_vm_fifo_update_done_irq
[ 1443.026082] pm8916_bms_vm_fifo_update_done_irq
[ 1496.532964] pm8916_bms_vm_fifo_update_done_irq
```
@TravMurav

aa889788, 2023-05-16:
> This hardware block is broken in many more ways than this, I remember seeing it timing wrong but decided it works good enough in any case given we can only do so much with it.
> 
> Not like I can wave a magic wand and fix a hardware bug in all of those chips in existence anyway...

I want to implement a daemon to update the bat LEDs when capacity and status(charging discharging) changed, but the bms trigger the uevent only every 50s, so is there any possibilities to trigger the bms changed event when charger online status changed?

TravMurav, 2023-05-16:
> I want to implement a daemon to update the bat LEDs when capacity and status(charging discharging) changed, but the bms trigger the uevent only every 50s, so is there any possibilities to trigger [...]

For this you should track your charger, not fuel-gauge.
Also FWIW there is a led trigger that enables the  led when device is charging

aa889788, 2023-05-16:
> For this you should track your charger, not fuel-gauge.
> Also FWIW there is a led trigger that enables the  led when device is charging

There are 4 LEDs to show the capacity and status, no suitable triggers for it, maybe I could listen the LBC online status and read the bms/uevent to get the status

TravMurav, 2023-05-16:
> > > > This hardware block is broken in many more ways than this, I remember seeing it timing wrong but decided it works good enough in any case given we can only do so much with it.
> > > > Not like I can wave a magic wand and fix a hardware bug in all of those chips in existence anyway...
> > > 
> > > 
> > > I want to implement a daemon to update the bat LEDs when capacity and status(charging discharging) changed, but the bms trigger the uevent only every 50s, so is there any possibilities to trigger the bms changed event when charger online status changed?
> > 
> > 
> > For this you should track your charger, not fuel-gauge.
> > Also FWIW there is a led trigger that enables the  led when device is charging
> 
> There are 4 LEDs to show the capacity and status, no suitable triggers for it, maybe I could listen the LBC online event and read the bms/uevent to get the status

Yes, if you want to track the online status, you'd track the lbc property that should change immediately, and for the capacity you'd track the bms update event