msm8916-mainline/linux #315: Asus z00l: Backlight doesn't turn on

Ro-Den, 2023-05-22:
In my device (ASUS ZenFone 2 Laser ZE550KL aka z00l) back light refuses to turn on at all.

I've managed to turn it on once by
`$ echo 50 > /sys/class/backlight/1a98000.dsi.0/brightness`
`$ echo 250 > /sys/class/backlight/1a98000.dsi.0/brightness`

The issue persists on at least 3 user interfaces I've tested: Phosh (Wayland), Sxmo-de-sway (Wayland), Sxmo-de-dwm (X11).

Adding a udev rule as proposed by @Jakko3 in #268 has not resolved the issue.

There are no significant errors in `dmesg` output:
> [    1.601106] msm_mdp 1a01000.display-controller: [drm:adreno_request_fw [msm]] *ERROR* failed to load a300_pm4.fw
> [    5.123357] l13: voltage operation not allowed
> [   21.692970] msm_mdp 1a01000.display-controller: [drm:adreno_request_fw [msm]] loaded qcom/a300_pm4.fw from new location
> [   21.693420] msm_mdp 1a01000.display-controller: [drm:adreno_request_fw [msm]] loaded qcom/a300_pfp.fw from new location

P.S.: Unfortunately, my device is a little bit broken with the Power Button completely ripped off. But I don't think it has something to do with the issue ):

stephan-gh, 2023-05-23:
There have been several reports of this issue but so far I have not been able to figure out why this happens. And I don't have this device myself.

It would be helpful if we knew if the display actually ever worked properly (perhaps there is some device-specific difference and it works only on some models like @antony-jr's?). It may help to build e.g. the good old [v5.11.7-msm8916](https://github.com/msm8916-mainline/linux/releases/tag/v5.11.7-msm8916) tag again and see if it works there.

So far investigating this issue always failed because the reporters disappeared eventually or where not willing to go through the effort of building the kernel repeatedly to debug this issue. Someone needs to step up if we want to solve this properly.

If you need help with the build process etc please join the chat at [#msm8916:postmarketos.org](https://matrix.to/#/#msm8916:postmarketos.org) on Matrix since it's easier to help there.

antony-jr, 2023-05-23:
My device display and battery is broken, but I've been meaning to repair and try the latest version of the kernel myself. But this is going to take some time to do the repairs. Give me a week or two, but can't promise anything.

Ro-Den, 2023-05-23:
> It would be helpful if we knew if the display actually ever worked properly

No, it never did in pmOS. There were no problems (except GPS) in Android though my device is not in a good shape ):

> perhaps there is some device-specific difference and it works only on some models like @antony-jr's?

As far as I know Asus produced 2 models of ASUS ZenFone 2 Laser ZE550KL: z00lDD (for Southeast Asia: Malaysia, Indonesia etc) and z00lD (for the rest of the world).

My device is z00lD. @antony-jr what's yours?

Ro-Den, 2023-05-23:
> Does TWRP work?

Yes, of course.

lk2nd works, too. Except for fastboot commands.

When my device boots, there's the Asus logo for a couple of seconds, then a grey screen for about a second, then boom - back light goes off. Is it normal behavior? Maybe, some over voltage protection steps in?

vknecht, 2023-05-30:
Been having that sort of problems for quite some time with AOSP, worked-around with https://github.com/vknecht/msm8916-mainline-linux/commit/ceba76ec3e5fbc02cf0a49ad02b4bebf133848ff

There was a discussion
https://lore.kernel.org/linux-arm-msm/CAA8EJpq_9iC1rkiZVom28Kv_B3QLd4pBgFObxBfSpJ+Xh=Mp1g@mail.gmail.com/
where it was suggested to split .prepare() / .unprepare() by using new .enable() /.disable() callbacks introduced by
https://lore.kernel.org/dri-devel/20221205173328.1395350-1-dave.stevenson@raspberrypi.com/

Didn't get to the bottom of it, maybe that should be done in panel generator
if that works, and maybe could help with not having to do this revert https://github.com/msm8916-mainline/linux/commit/7f74040bd869f69da0282720bccee845f3224da6

Ro-Den, 2023-07-05:
I have finally tested a number of older kernels.

First, I tried to build the good old [v5.11.7-msm8916](https://github.com/msm8916-mainline/linux/releases/tag/v5.11.7-msm8916) myself, but encountered a lot of issues. After that I switched to the relevant branch, added and built my device, but still could not get back light working. Then I tried a number of newer kernels same way, but the issue persisted.

So, it's definitely either the faulty device or some hardware mismatch.

knuxify, 2023-08-03:
I don't own this device, but I checked the schematics out of curiosity, and it seems that this phone is most likely using the RT4532 chip for backlight control: [manufacturer website](https://www.richtek.com/Products/LED%20Driver/backlight-led/RT4532.aspx), [datasheet](https://www.richtek.com/assets/product_file/RT4532/DS4532-01.pdf). Depending on how it's connected to the SoC, it may be possible to drive it separately from the panel driver, which might solve some of the issues.

The site mentions PWM control; I'd start by checking if we can't use `pwm-backlight` to control the backlight, then move on to figuring out if a separate driver isn't necessary.

Ro-Den, 2023-11-21:
The back light issue is solved in kernel 6.6.0

But now `$ echo 50 > /sys/class/backlight/1a98000.dsi.0/brightness` returns `Access denied`. And changing brightness level requires pressing the Power button.

closed 2023-11-21 by Ro-Den

Jakko3, 2023-11-22:
The pmOS wiki page on LEDs contains some possibilities on the permissions in sysfs. This is mainly useful if you want to change it by command line or by script.
- https://wiki.postmarketos.org/wiki/LEDs#Access_via_sysfs

For backlight, however, actually it shouldn't be needed to handle this by command line or script. The brightness control of the user interface (desktop environment) should be able to manage it. In case you want to change something at boot, an udev rule like mentioned in #268 would work without changing persmissions. I'm not sure what you mean by pressing the power button, though.

Ro-Den, 2023-11-22:
Thanks!

`$ echo 50 | sudo tee /sys/class/backlight/1a98000.dsi.0/brightness` works!

Anyway, I have to press the Power button or `xdotool key 122` to turn off the display for the changes to come into effect.

Jakko3, 2023-11-22:
> Anyway, I have to press the Power button or `xdotool key 122` to turn off the display for the changes to come into effect.

Ok, thanks for explaining. That seems to be something driver-specific.