msm8916-mainline/linux #335: samsung j5: pixel rain

chr-chr, 2023-10-23:
Hi,

The display works good, but ...
I get a pixel rain on certains colors, e.g. it doesn't like #d4d4d4

This also happens on posh/wayland.

Any help appreciated

[image]

wonderfulShrineMaidenOfParadise, 2023-10-25:
It's a known issue, but the cause is unknown. I almost forgot it happens.

Some clues about the panel driver:
- https://github.com/msm8916-mainline/linux-downstream/blob/SM-J510FN/drivers/video/msm/mdss/samsung/S6E8AA5X01_AMS520KT01/dsi_panel_S6E8AA5X01_AMS520KT01_hd_octa_video.dtsi#L63-L64
- qcom,mdss-dsi-bllp-eof-power-mode;
- qcom,mdss-dsi-bllp-power-mode;

Perhaps we need some sort of flags for these 2 modes, but I don't even understand what they are.

chr-chr, 2023-10-26:
Well,

I'm already hacking on this, my devices uses "ams497hy01":

https://github.com/msm8916-mainline/linux/pull/333

However, I almost got the assumption this is broken HW since the displays do have some cracks.
But ... the rain modulates with system load and not by applying mechanical force.

I had 2 devices in a row having this glitch. On posh hardly to notice since it doesn't use that colour. But uploading a screenshot of the mate desktop shows the same glitches.

Meanwhile I flashed another 2 devices with PMOS mate/posh: flawless.

And I reflashed one of the pixel rain back to lineage: flawless !

And one devices, lineage ok but with PMOS just a flickering screen ...

Actually I assume all my devices are the same type and I updated baseband fw on all of them. Going to double check if these devices differs.

wonderfulShrineMaidenOfParadise, 2023-10-27:
Currently it's confirmed on both J5(X), with unscratched panels.
Fun fact: The glitch pattern varies on different phones, even on the same variant of J5X.