msm8916-mainline/linux #339: consider changing paths for WCNSS_qcom_wlan_nv.bin

jenneron, 2023-11-07:
I'm looking into implementing a generic port for these devices, and I'm facing this problem: we have this file in the same path on all devices, so we can't install all of them at the same time.

It can be fixed by changing firmware path and according to https://github.com/torvalds/linux/blob/be3ca57cfb777ad820c6659d52e60bbdd36bf5ff/drivers/soc/qcom/wcnss_ctrl.c#L213 it can be done in device-tree on per-device basis

Another way could be to create a symlink with msm-firmware-loader. I'm not sure which way you would prefer

TravMurav, 2023-11-07:
Yes, this is one of the not-yet-solved things we need to do...

Unfortunately we can't reliably use msm-fw-loader since some (many) devices keep this config in /system, so we have to package them. For the select few that contain it in /persist, the fw-loader has code to handle it: https://gitlab.com/postmarketOS/msm-firmware-loader/-/blob/master/msm-firmware-loader.sh?ref_type=heads#L109-113

One of the ides we had is to collect all the nv files, name them per-board and create a big dump repo to install in the os, then add firmware-name to all the devices dts. Unfortunately we still had few open questions (like who would host this archive, what licensing considerations are there for distributing those files...) and so it was on a back burner for a while now...

TravMurav, 2023-11-07:
(cont. after I accidentally pressed ctrl+enter)

FWIW the devices seem to work semi-fine even using the sbc's nv file. This is, theoretically, wrong due to the different antenna calibration, but doesn't, for example, prevent 5GHz devices from working properly...

If you need a stop-gap for testing, you could probably use that fine.