msm8916-mainline/linux #342: samsung-a5lte: microphone does not work during calls in msm8916/6.6 and newer

Grimler91, 2024-02-04:
With a kernel built from msm8916/6.6 microphone does not work during calls (at least in phosh), the person on the other end of the call hear nothing.
With a kernel built from msm8916/6.5.2 the microphone works fine in calls.

I have tracked the issue down to commit ["FIXME: ASoC: qcom: msm8916: enable jack detection kctls (v2)"](https://github.com/msm8916-mainline/linux/commit/baad968b89b877f31f4e909a6bc599188a4bb58a), which was dropped in msm8916/6.6 branch. 
Simply applying that commit on top of 6.6 or 6.7 or newer makes the microphone work again.

TravMurav, 2024-02-05:
Hi! Can you check that https://gitlab.com/postmarketOS/pmaports/-/merge_requests/4787 fixes the issue?

Grimler91, 2024-02-05:
@TravMurav it works indeed, thanks!

closed 2024-02-05 by Grimler91