msm8916-mainline/linux #347: Unable to find xxxx.dtb in the following locations, I need help,

kexirong, 2024-03-14:
I built with reference to this [document,](https://wiki.postmarketos.org/wiki/MSM8916_Mainlining) but it reported an error that dtb could not be found.
I haven't been able to resolve this error so I need help
My configuration related files:
[APKBUILD.txt](https://github.com/msm8916-mainline/linux/files/14596422/APKBUILD.txt)
[deviceinfo.txt](https://github.com/msm8916-mainline/linux/files/14596423/deviceinfo.txt)
[msm8916_defconfig.txt](https://github.com/msm8916-mainline/linux/files/14596424/msm8916_defconfig.txt)
I have added the dts file and fixed the mikefile:
[image]
[image]
My command steps:
[image]
pmbootstrap install error message:

[image]

TravMurav, 2024-03-14:
Hi, by old convention 32bit arm platforms put all dtb files into the root of dtbs dir so you don't need leading `qcom/` dir in the path, this is what `qcom-` prefix in the file is for. However do you really need to build 32bit kernel for this device?

kexirong, 2024-03-14:
@TravMurav Thanks, I will try to remove qcom and rebuild. Yes, my hardware only supports 32 bit, I tried flashing its tz partition to support 64 bit, but that gave me quite a few weird problems that made me give up on 64 bit, so I tried building a 32 bit postmartkos

kexirong, 2024-03-14:
@TravMurav  Hello, the problem still exists. Thank you. I feel that the makefile of armv7 will not compile the dts file starting from 6.5.2. I will try 6.3.
[image]

closed 2024-03-14 by kexirong

kexirong, 2024-03-14:
add CONFIG_ARCH_QCOM=y in msm8916_defconfig

vi pmbootstrap/cache_git/pmaports/device/community/linux-postmarketos-qcom
-msm8916/APKBUILD  change /boot/dtbs to  /boot/dts