msm8916-mainline/linux #369: Touchscreen zinitix fails with "too many keys defined"

Jakko3, 2024-09-30:
@TravMurav: On device samsung-serranove the branch `wip/msm8916/6.9` runs fine but on `wip/msm8916/6.10` and `wip/msm8916/6.11` the touchscreen fails with message "Zinitix-TS 5-0020: too many keys defined (-22)" followed by "Zinitix-TS 5-0020: probe with driver Zinitix-TS failed with error -22". Looks like the last part of https://github.com/msm8916-mainline/linux/commit/d3982377cbca3f1eaa829f9a1eabbc25ba59de6a is causing this.

The number in the brackets after "too many keys defined" returns the value of `bt541->num_keycodes` as an integer. The value comes from `device_property_count_u32()`, which is a modified [1] form of `device_property_read_u32_array()`. This function [2] is supposed to return the "number of values" (in our case that's the number of keycodes) if argument "val" is set NULL, which the modified function `device_property_count_u32()` does. However, there is also a list of possible error return codes.

On device samsung-serranove, the touchkeys below the screen are handled by driver "tm2-touchkey" [3] being called via devicetree match "coreriver,tc360-touchkey". Therefore I assume that the samsung-serranove zinitix touchscreen doesn't contain a property "linux,keycodes".

In the above mentioned list of possible error return codes I would have thought that `-ENODATA if the property does not have a value` would be the return value. Though I guess this would be the case only if there is a property "linux,keycodes" but it doesn't contain any data. The return value of "too many keys defined" is -22, which is `-EINVAL` [4]. It the case of `device_property_read_u32_array()` that means `-EINVAL if given arguments are not valid`, which is a bit cryptic but seems to apply if there is no property "linux,keycodes" at all.

The error message "too many keys defined" is triggered by the comparison `if (bt541->num_keycodes > ARRAY_SIZE(bt541->keycodes))`. At first it seems a bit strange why the first value -22 should be bigger than the second value, which to my understanding should be 8. The type of `bt541->num_keycodes` is integer but I'm not sure about the type of ARRAY_SIZE(), this is possibly an unsigned integer. In that case the comparison would be performed for two unsigned integers [5], turning -22 into a large positive number.

Long story short, I think the case of "linux,keycodes" property not being available isn't covered.

- [1] https://elixir.bootlin.com/linux/v6.11/source/include/linux/property.h#L250-L253
- [2] https://elixir.bootlin.com/linux/v6.11/source/drivers/base/property.c#L129-L154
- [3] https://elixir.bootlin.com/linux/v6.11/source/drivers/input/keyboard/tm2-touchkey.c
- [4] https://elixir.bootlin.com/linux/v6.11/source/include/uapi/asm-generic/errno-base.h#L26
- [5] https://stackoverflow.com/questions/35079300/comparing-negative-numbers-in-c

TravMurav, 2024-10-01:
Gah! seems like I've introduced this mistake at the last moment when fixing the patch per upstream review... Thanks for noticing it! I've prepared a fix here: https://github.com/msm8916-mainline/linux/commit/f5ce53208208b3add4992de23d9f70cb55a12698

Please check that it fixes the issue and that you're fine with the `Reported-by:` I've put there and I will submit it upstream in few days

closed 2024-10-01 by Jakko3

Jakko3, 2024-10-04:
I also tested v2. Works fine as well.

https://lore.kernel.org/linux-input/20241004-zinitix-no-keycodes-v2-1-876dc9fea4b6@trvn.ru