msm8916-mainline/linux #388: Should PSCI configuration be removed from `msm8916.dtsi`

samcday, 2025-01-22:
After speaking with @TravMurav and reading the https://github.com/msm8916-mainline/lk2nd/commit/8183ea211bd68a6e1342826e78265d37eedb8899 commit message from @stephan-gh in lk2nd , my understanding is that essentially every device that ever shipped with an msm8916 SoC, did so *without* a TZ firmware that implements PSCI. Apparently there was a firmware update that came later for db410c that implemented PSCI, but it was buggy and/or not really usable.

Have I summarized that history accurately?

If so, I'd like to propose that we yeet all mentions of PSCI from `msm8196.dtsi`:

 * Remove the `/psci` node entirely, since `lk2nd` just disables it anyway: https://github.com/msm8916-mainline/lk2nd/commit/8183ea211bd68a6e1342826e78265d37eedb8899#diff-fd8a4c67bcf754ee8cf1231f406048b5eebf4fe1fe3c4624e7c501ac53319f25R237
 * Change the `method` property of the `/cpu` nodes to `spin-table`.

If it turns out there are some devices that *did* ship with a working PSCI implementation that can actually be used to bring secondary cores online, we could introduce a `msm8916-psci.dtsi` overlay to address those exceptions to the rule.

The reasons I think this work would be advantageous:

 * Ensures that the upstream device trees for msm8916 actually reflect reality.
 * Removes a source of confusion for folks studying these device trees to understand this platform..
 * Makes it easier to add support for bringing the secondary cores online in U-Boot, since it already [has support](https://github.com/u-boot/u-boot/blob/ebc84d7/arch/arm/cpu/armv8/Kconfig#L44) for doing most of the grunt-work to bring the cores online (it sets up the `cpu-release-addr` with the [`wfe` busy-loop](https://github.com/u-boot/u-boot/blob/ebc84d7b60c1ed3398e9f600fe3dc8406500bd35/arch/arm/cpu/armv8/spin_table_v8.S) instructions, and [patches the tree](https://github.com/u-boot/u-boot/blob/ebc84d7b60c1ed3398e9f600fe3dc8406500bd35/arch/arm/cpu/armv8/spin_table.c#L10)).

samcday, 2025-01-25:
Another reason to do this is it makes it easier/possible to have proper sysreset support in the generic `qcom_defconfig` config in upstream U-Boot: https://lore.kernel.org/u-boot/20250125-msm8916-sysreset-v1-0-62073932ff0e@samcday.com/T/

stephan-gh, 2025-01-26:
> After speaking with @TravMurav and reading the https://github.com/msm8916-mainline/lk2nd/commit/8183ea211bd68a6e1342826e78265d37eedb8899 commit message from @stephan-gh in lk2nd , my understanding [...]

This is correct until here.

> Apparently there was a firmware update that came later for db410c that implemented PSCI, but it was buggy and/or not really usable.

That's a bit of a stretch. I did find out that the PSCI firmware is buggy while working on some WIP patches meant to improve power efficiency during suspend. But that's not something you can hit with upstream Linux today. PSCI does bring up secondary CPU cores just fine. In other words, everyone deploying a DB410c nowadays likely uses the PSCI firmware without noticing any problems. We can't just remove PSCI support entirely for MSM8916, as it would certainly break most users of DB410c.

I have mixed feelings about making this change upstream. There are definitely arguments in favor of doing that. It's certainly more accurate for the majority of devices. But it doesn't _have_ to be true. For example, I do use PSCI on some of my devices (e.g. bq-paella). The firmware for DB410c works just fine on smartphones as well, assuming they do not have firmware secure boot enabled. This is why lk2nd implements the dynamic detection at runtime. Some people will have the original firmware flashed and others perhaps the PSCI one.

The problem with `spin-table` is that it only allows booting the secondary CPU cores once. For power saving, you need to dynamically turn CPUs off and back on (cpuidle). Officially, upstream arm64 Linux only supports this with PSCI. No other implementations are allowed, because the goal was to force vendors to adopt PSCI. There is a [partial workaround for that downstream here in msm8916-mainline/linux](https://github.com/msm8916-mainline/linux/commit/36baf027e62fa1ef51998f483d680cf26c9d9318), but it's incomplete, and upstream doesn't have any of that.

To still get power saving in upstream, I've experimented with different options for retrofitting MSM8916 with proper (non-buggy) PSCI support. The [Trusted Firmware-A port for MSM8916](https://trustedfirmware-a.readthedocs.io/en/latest/plat/qti-msm8916.html) is a (partial) open-source PSCI implementation for MSM8916. It works primarily as EL3/`tz` replacement for devices without secure boot, but I've also [experimented with running it as EL2/`hyp` replacement](https://github.com/msm8916-mainline/arm-trusted-firmware/commit/08a93cf139d7df59c8979b99786a4cf0256fcd6d). The latter could potentially work on all ARM64-capable MSM8916 devices.

The only alternative to get power saving without PSCI would be to convince upstream to allow other cpuidle drivers on ARM64. I've also considered doing that, but it will certainly trigger some controversies.

It's a complex topic with several different implementation options. The fact that upstream describes everything as PSCI-capable has mostly historical reasons, but I also couldn't convince myself to change it, to avoid making myself more work later if we would decide to go e.g. with the PSCI-in-EL2 approach.

samcday, 2025-01-27:
Thanks for the in-depth explanation and clarifications!

> That's a bit of a stretch. I did find out that the PSCI firmware is buggy while working on some WIP patches meant to improve power efficiency during suspend.

After I wrote this original issue, I stumbled across [some of your comments in U-Boot](https://github.com/u-boot/u-boot/blob/master/arch/arm/mach-snapdragon/include/mach/boot0.h#L9), which I'll quote here:

 > * The PSCI implementation in the TrustZone/tz firmware on DragonBoard 410c has
 > * a bug that starts all other CPU cores in 32-bit mode unless the TZ syscall
 > * that switches from 32-bit to 64-bit mode is executed at least once.

Would that be the main issue or was there others?

>  We can't just remove PSCI support entirely for MSM8916, as it would certainly break most users of DB410c.

Right, so I'll update my original issue description in a moment, but I guess the proposal would be "*move* the `/psci` node and related PSCI properties from `msm8916.dtsi` to `msm8916-psci.dtsi` and make sure db410c depends on that".

> To still get power saving in upstream, I've experimented with different options for retrofitting MSM8916 with proper (non-buggy) PSCI support.

Very cool! I'd love to experiment with that stuff more, but my current arsenal of msm8916 devices are all Samsungs, and the [edl tool doesn't seem to work so well with these](https://github.com/bkerler/edl/issues/525), so I'm too scared to mess with SBL (or even aboot 😅)

> The only alternative to get power saving without PSCI would be to convince upstream to allow other cpuidle drivers on ARM64. I've also considered doing that, but it will certainly trigger some controversies.

Yeah, I certainly wouldn't want to open that can of worms, and I think it's a tough case to make when we'd be proposing to do it mostly to support a decade-old platform.

> but I also couldn't convince myself to change it, to avoid making myself more work later ...

This is very reasonable. I wouldn't want to create more work for you either! That's why I was offering to do this work :)

> ... if we would decide to go e.g. with the PSCI-in-EL2 approach.

I think this is the sticking point for me. It seems to me like we may still be

closed 2025-01-27 by samcday

stephan-gh, 2025-01-28:
> Thanks for the in-depth explanation and clarifications!
>
>
> After I wrote this original issue, I stumbled across [some of your comments in [...]

That's also buggy, but it's not the bug I meant here. That workaround in U-Boot works pretty well, so I didn't even think about that anymore.

The issue I meant happens if one tries going to the lowest power state. It doesn't power off the CPU power supply correctly, which seems to cause some leakage or similar. Basically, it ends up consuming more power than during runtime. 😅 But that's quite far away on mainline, there are lots of changes needed to even reach that lowest power state.

> > We can't just remove PSCI support entirely for MSM8916, as it would certainly break most users of DB410c.
> 
> Right, so if I had the complete story at the time I opened the issue, I'd have worded the proposal along the lines of "_move_ the `/psci` node and related PSCI properties from `msm8916.dtsi` to `msm8916-psci.dtsi` and make sure db410c depends on that".
> 

That's probably okay. Note that the inverse of that already exists in mainline for ARM32: `arch/arm/boot/dts/qcom/qcom-msm8916-smp.dtsi`. Unlike ARM64, non-PSCI is fully supported on ARM32, that's why it's already used there. (Also, ARM32 Linux doesn't support spin-table...)

> > To still get power saving in upstream, I've experimented with different options for retrofitting MSM8916 with proper (non-buggy) PSCI support.
> 
> Very cool! I'd love to experiment with that stuff more, but my current arsenal of msm8916 devices are all Samsungs, and the [edl tool doesn't seem to work so well with these](https://github.com/bkerler/edl/issues/525), so I'm too scared to mess with aboot 😅, let alone TZ/SBL (or actually, EDL lives on SBL right? how would one even recover from a bricked TZ/SBL in that case?)
> 

You can't replace any firmware on the Samsung because it has firmware secure boot enabled. The only way you can boot something in EL2 is via [CVE-2022-22063](https://github.com/msm8916-mainline/CVE-2022-22063), which can be implemented in lk2nd or U-Boot to replace the loaded hypervisor firmware at runtime. The legacy branch of lk2nd implements something similar already, we didn't port it to the new lk2nd branch yet. Since this happens all at runtime, there is no need to worry about recovery. :-)

The default hypervisor doesn't seem to do anything useful, so it would be a convenient place to reuse for a proper PSCI implementation. The proof of concept in https://github.com/msm8916-mainline/arm-trusted-firmware/commit/08a93cf139d7df59c8979b99786a4cf0256fcd6d does work well enough for booting CPU cores already, so in theory it's not years away. It does lack support for deep suspend/idle states, but the effort required to implement that is likely less than upstreaming all the required non-PSCI code to Linux...

Most of this is only relevant when implement support for deep low power suspend states though, and the details of that are so complicated that this might indeed still be years or forever away. 😅

samcday, 2025-01-29:
> The issue I meant happens if one tries going to the lowest power state. It doesn't power off the CPU power supply correctly, which seems to cause some leakage or similar. Basically, it ends up consuming more power than during runtime.

Oops! 🙈 

> You can't replace any firmware on the Samsung because it has firmware secure boot enabled. The only way you can boot something in EL2 is via [...]

Ah-ha, it all makes sense now. I remember looking at the CVE-2022-22063 early last year but my general understanding of armv8/snapdragon architecture was too limited to properly understand what I was looking at. What a fantastic writeup, by the way ⭐ 

> The default hypervisor doesn't seem to do anything useful, so it would be a convenient place to reuse for a proper PSCI implementation. The proof of concept in [...]

That does indeed seem like a promising and viable approach. When I find some free cycles I'll experiment with this.

>  It does lack support for deep suspend/idle states, but the effort required to implement that is likely less than upstreaming all the required non-PSCI code to Linux...

For now I've sent a [patch upstream to U-Boot that detects when PSCI is absent and neuters the `/psci` node](https://patchwork.ozlabs.org/project/uboot/patch/20250127-qcom-handle-absent-psci-v1-1-e762f2db938c@samcday.com/), as well as a patch series that [does the spin-table SMP bring-up](https://patchwork.ozlabs.org/project/uboot/list/?series=441966) similar to how lk2nd / arm32-kernel does it.