msm8916-mainline/linux #395: Update to 6.12.19

sergiotarxz, 2025-03-20:
I have discovered that doing this in the 6.12.1 branch:

```shell
git remote add upstream git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git
git config pull.rebase false
git pull upstream v6.12.19
```

Results in a buildable kernel with no merge conflict.

Would you accept a pull request to create a v6.12.19 branch?

TravMurav, 2025-03-20:
Hi! Development in this repo generally follows mainline linux, mostly using "wip" branches (most current one being `wip/msm8916/6.14-rc7`). The rare exception to this rule is if we're late with a release and a stable branch exists for a version we're making a release for, which was the case for `v6.12.1-msm8916` release tag.

Since it generally doesn't make sense to do development on stable/lts branches, I don't see the reason to carry them here. If one wishes, they can build their own kernel by rebasing the relevant patchset locally to whichever version they wish.

sergiotarxz, 2025-03-20:
Hi @TravMurav 

This is sad and dangerous, but ultimately not your fault as a project at all.

https://gitlab.postmarketos.org/postmarketOS/pmaports/-/blob/master/device/community/linux-postmarketos-qcom-msm8916/APKBUILD?ref_type=heads

You are the upstream of postmarketos for the msm8916 devices, any security fixes for Linux vulns have to wait to a new stable version being released by yourselves or they must pick rcx versions.

Anyway not your fault, feel free to close.

TravMurav, 2025-03-20:
Unfortunately we really don't have the capacity to /test/ each new stable tag (or even each mainline rc, as evident by the amount of non-wip/tagged branches compared to wip branches that have minimal to none qa) and experience shows that down the line random backported "fixes" or their random automatically pulled dependencies actually break things for us, requiring long bisects and debugging effort. In case of some major problem affecting security though we'd prepare a tag with fixes backported, something we already did once when a major problem in usb stack was found.

Still thanks for bringing this up, I appreciate you taking your time to look at the project

closed 2025-03-20 by TravMurav