msm8916-mainline/linux #396: Issues with incoming SMS and SIM-cards

Ro-Den, 2025-04-13:
1. I've recently encountered a problem with receiving SMS on one mobile operator's network.

The problem appeared within a year and it persists in different regions.

Outgoing SMS is being delivered without problems.

I've checked 2 other operators and found no issues in receiving SMS.

In Android there's no SMS issues with all operators.

2. SIM-cards of 4th mobile operator don't work at all.

The operator allows the usage of SIM-cards with cheap tariff plans only in smartphones and blocks them in non-telephone devices.

As I understand, the modem's firmware was taken from some non-telephone device, so nothing can be done in this case. Please, correct me if I'm mistaken.

TravMurav, 2025-04-13:
Hi! I'm afraid that modem issues aren't caused by the kernel most of the time, instead it's userspace software stack (i.e. ModemManager) that talks to the device-specific modem firmware, which linux only boots.

For the second question, it's most likely that your operator checks IMEI of your device, nothing kernel can do in this case either I'm afraid (fwiw I don't even know what device or distro are you using  to give any tips...)

Ro-Den, 2025-04-13:
> it's userspace software stack (i.e. ModemManager) that talks to the device-specific modem firmware, which linux only boots.

But the user-space software has not changed. I installed pmOS on samsung-j5x and samsung-a5 a year ago and everything was fine, except that every SMS doubled ); Two months ago I upgraded pmOS on samsung-a5 and stumbled upon this issue. Then I checked samsung-j5x which was not upgraded (though, I've reinstalled Plasma Dialer; and Modem Manager GUI segfaults now; need to investigate it further ); and got the same issue, too. If it's SIM-card then Android would have the same issue, too. But it doesn't. I guess they've changed something on operator's side that causes the issue somewhere in Linux.

> it's most likely that your operator checks IMEI of your device

But installing pmOS doesn't change the IMEI, does it? So, it's a little bit more complicated. All I know is that I had to flash some special firmware in my LTE router to be compatible with the SIM-cards of this goddamn mobile operator. Once I switched to a cheaper tariff plan their SIM-card stopped working in the LTE router and in pmOS on both Samsungs. But of course it continues to work in Android.